armel/M5HamRadio #1: ini file not being read

G0MJW, 2022-10-13:
I can't get this to work. It just fails to read the ini file. I initially thought it was the 32GB card but  I tried another, also formatted to fat32 that was only 2 Gb and the same. This is on the M5STACK Basic. Perhaps it is not up to the job?

armel, 2022-10-13:
Hi,

Ok. 
What do you see on the M5Stack screen? Is there a message?
Can you post or send me your ini file?

73'

Armel F4HWN.

G0MJW, 2022-10-13:
Hi Armel,

 

I don’t see anything, just failed to load config file.

 

It’s the same as your example

 

; Icom Config

[icom]

icom_model = 705

icom_address = 0xA4

icom_connect = BT

icom_device = 30:31:7D:33:B2:58

 

; Wifi Config

[wifi]

wifi_ssid = mjw

wifi_password = XXXX

 

; Proxy Config

[proxy]

baud_rate = 115200

proxy_url = http://192.168.1.32

proxy_port = 1234

 

; Transverter Config

[transverter]

transverter_lo_1 = 116000000

transverter_lo_2 = 118000000

transverter_lo_3 = 404000000

transverter_lo_4 = 406000000

transverter_lo_5 = 9968000000

 

; Geolocation Config

[geolocation]

latitude = 48.848285

longitude = 2.2708201

armel, 2022-10-13:
Humm, something is not working.
Could you, please, take a look at this

https://github.com/armel/M5HamRadio#help-me-to-debug-your-problem

And send me the trace log ?

Armel F4HWN.

G0MJW, 2022-10-13:
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1184

load:0x40078000,len:13160

load:0x40080400,len:3036

entry 0x400805e4

E (439) psram: PSRAM ID read error: 0xffffffff

SPIFFS/DXTracker.bin

SPIFFS/ICSMeter.bin

SPIFFS/ICMultiMeter.bin

----------

[  2037][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work

----------

[  6272][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work

Loading config file...

/.backup.ini

Ini file does not exist.

 

https://github.com/armel/M5HamRadio#help-me-to-debug-your-problem

armel, 2022-10-13:
Okay, well done, thank you very much.
This is very useful.
It seems that your M5Stack is not able to mount  the SD Card. 

**sdcard_mount(): f_mount failed: (3) The physical drive cannot work**

So your M5Stack is not able to read a valid .ini file on it. I'm using 4, 8, 16 or 32 Gb SD Card, in FAT32, without any problem. Could you confirm there is no partition on your SD Card ? 

Armel F4HWN.

G0MJW, 2022-10-13:
I tried two, I reformatted both to just one FAT32 partition. One was a new 32 Gb Sandisk ultra, the other a very old 2 Gb card. It can’t read either. I have ordered a 16Gb sandisk card just in case.

armel, 2022-10-13:
OK Michael. Let's wait for the new card. Please also check if your SD card is inserted correctly (open your M5Stack if necessary). The message that appears is a low-level system message that is not from my code. I have only seen it in the past because of a bad formatting, a partition problem or defect on the card. I hope it's not a problem with the onboard SD card reader :(

Armel F4HWN.

G0MJW, 2022-10-14:
I won’t see the new card until next week, but it is not really any different than the 32Gb one, except it will be a virgin and not reformatted. I thought I had some more lower capacity cards but it appears not. I have bought many in the past, I wonder where they all go, they are like odd socks, just vanish. It was inserted correctly, I checked that.

 

Meanwhile I can see I might need to compile this and install, then I would not need a SD card at all. In fact, to just connect to an IC705 via Bluetooth presumably no Wifi access is necessary. I will need to install the tool chain to see if these modules are useful, I have also bought a communications stack with RS422, CAN bus etc. A larger screen variant might do a good job controlling a linear amplifier or managing antenna switching.

 

Mike

G0MJW, 2022-10-14:
Interestingly I just tried the 32 Gb card with another firmware that sniffs WiFi packets and saves them on the SD card. This worked fine. I am wondering if there is more than one type of these units? 

 

Mike

 

https://github.com/armel/M5HamRadio#help-me-to-debug-your-problem

G0MJW, 2022-10-18:
It now doesn't appear to be able to connect. I have given up for now.

armel, 2022-10-18:
Hi,

First of all, if you are using Bluetooth with an IC-705, you need to set the address of your IC-705 correctly in your .ini settings file. For me, icom_device = 30:31:7D:33:B2:58. But yours is probably different. 

To get the address of your IC-705, from your transceiver, enter Menu and go to Set > Bluetooth Set > Bluetooth Device Information.

After that, you need to perform the first manual pairing. You need to connect your IC-705 to your M5Stack. To do this, consult the Icom documentation, if necessary. This is just like pairing any BT device.

Armel.

G0MJW, 2022-10-18:
Ah, modifying that device address is not mentioned in the instructions, nor in any of the online videos I watched, but the general flakiness of my M5stck to me looks like faulty hardware. Most probably in the PSU circuitry. Odd it read the SD card once but not twice. Boot timing maybe? By the way it seems to not default as it should, I had to press the button during the greed load to get the red or blue. Also, when burning it sits at 26% for a long time before completing. This points to faulty memory perhaps?

 

Mike

armel, 2022-10-19:
According to the README, _take the time to adapt the values to your use (Icom CI-V Address, Wifi SSID and password, and so on). It's important !_ 🤔 Writing comprehensive documentation is difficult.

Armel.

G0MJW, 2022-10-19:
It doesn’t mention that device address. Anyway I changed it to the one for the radio. It now says

 

icom_device = 30:31:7D:50:AA:87 

 

but I got exactly the same result. It would not pair. I tried many times, I deleted the device, I reburned, I even reset the IC705 Bluetooth. It is detected, but when attempting to pair It asks for a passkey confirmation, which I accept, and then just times out with a ‘Connection failed’.

 

This is the same for both ICSMeter and ICSMultimeter.

 

So then at random, I decided to reboot and select the ini file again from the second loading menu. This time it rebooted again but connected! This must be a bug.  Do you recall I said it didn’t start up in the right mode when initially started. Could it be reading the DXTracker ini file for ICMeter and vice versa instead?

 

Given this I rebooted and it still connected, so I assume it is stored in flash. 

 

Then I tried DXTracker again. That just sits in a boot loop, but again, if I manually select the correct ini file from the menu in the second boot screen it loads and reboots, then it loads DX tracker.

 

So, I think there is a problem reading the ini files and it reads the wrong one. It does connect to WiFi so it must be reading something.

 

I suspect the device address is a red herring as nobody mentions it in any of the videos I have seen. Maybe the issue has arisen with the recent change to 3 firmwares in one?

 

Mike

armel, 2022-10-19:
Hi Mike,

Okay.

When you select an .ini file with the Ini Loader (and if it is valid), a copy is made on the Flash memory of the M5Stack. And if at the next reboot you do not make any action / selection of .ini file with the Ini Loader, it is this version in Flash memory that will be read automatically. This way, you can remove the SD card from your M5Stack, if you wish. 

So, if you change something in your .ini file (for example, the Wifi password, etc.), you must restart your M5Stack and select it again from the Ini Loader to force reload this new version. A new copy will then be made in the M5Stack Flash memory. And it will be this new version that will be read at the next reboot (if no action / selection with the Ini Loader).

About the videos, they are obsolete. They refer to version 1.X of these projects. The operating mode has changed since then, especially with the use of M5Burner and .ini files.

Armel F4HWN.

G0MJW, 2022-10-19:
I understand that. I think the problem is the wrong ini file being selected on first boot, thus requiring manual intervention.

closed 2022-12-20 by armel