armel/M5HamRadio #3: COM Port  Issue...

DaddyWarhammer, 2022-10-20:
Project requires COM port to communicate with M5Stack. 
My Win7 computer has no COM ports. 
Is there a virtual Serial port driver that will work?

DaddyWarhammer, 2022-10-20:
Problem is, when I install the drivers, Windows Device Manager flags the entry and gives an error 52.
[image]
After examining the certificate, it appears to be expired.
[image]

armel, 2022-10-20:
Ok, well, this is an operating system issue due to your Windows version and settings. This is not an issue about my projet. I don't work under Windows, but take a look at this link :

https://www.wintips.org/how-to-fix-windows-cannot-verify-the-digital-signature-for-this-file-error-in-windows-8-7-vista/

It explains how to bypass and disable the _Driver Signature Enforcement_. May be this tip works. Note that, Windows 7 support has been ended for almost 3 years. 

If this does not work, contact M5Stack. I can't help you install the M5Stack device. I don't develop drivers, just firmwares who work on M5Stack.

73'

Armel F4HWN.

DaddyWarhammer, 2022-10-20:
OK, thanks - I'll try that.
I think, ultimately, I'll need to upgrade to a new laptop and use the newer universal driver M5Stack has on their site.

Anyway, I do appreciate your quick replies and I can't wait to get the code running!
Regards,
Paul.

closed 2022-10-20 by DaddyWarhammer

armel, 2022-10-20:
Ok Paul. 
I'm sure you'll fix this problem. Driver issues are always a pain to solve, i know 😔
Let me know if you have any other problem. I will try to do my best.

73'

Armel F4HWN.

DaddyWarhammer, 2022-10-21:
Thank you, sir.