armel/M5HamRadio #4: Lost Touch on M5Core2

pa3ang, 2022-10-26:
After downloading and burning with M5Burner the display showed:
1. BinLoader V0.4
2. Failed to load config file. Please, add a valid config file on SPIFFS or SD Card

Now when loading the M5Core2_Factory test the buttons and Touch is lost. During startup Touch it is still recognised but does not work.  

Any idea?  

Bit disappointed and reluctant to burn the firmware on aaaa second M5Core2 and ending up with two failty units.

Johan, PA3ANG

pa3ang, 2022-10-26:
[FROM M5Burner] COM22 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
SPIFFS/DXTracker.bin
SPIFFS/ICSMeter.bin
SPIFFS/ICMultiMeter.bin
----------
SD/bootcode.bin
----------
----------
----------
Loading config file...
/.backup.ini
Ini file does not exist.

armel, 2022-10-26:
First of all, hello to you...

Well, I reassure you that my firmwares will not destroy or alter your M5Stack. Hundreds of OMs have been using it for months.

So if M5Core2_Factory is not working, there is maybe a problem with your M5Stack Core 2 unit. 

Last, but not least, please read the README.md file completely. Did you format an SD card in FAT32? Did you copy a valid .ini file to the root of this SD card (after adapting the values to your usage?). I think not. 

So please try again. Trust me, it works. It's just not plug and play and it never will be. I can't make up your settings. It's up to you to do it via the .ini file.

Armel F4HWN.

pa3ang, 2022-10-26:
Hi Armel,

Thanks for the fast response, really appriciated.

Because i was attracted to the M5Burner and  the M5HamRadioCore2 stack i did not pay attention to the Github README.md anymore. Sorry for that.  
I now have some .ini file on a SD card, but still getting the error message. This could now happen because i can't use the middle button to select a .ini file?  But also when i place one file in the SD root it will not start.

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4
SPIFFS/DXTracker.bin
SPIFFS/ICSMeter.bin
SPIFFS/ICMultiMeter.bin
----------
----------
----------
SD/DXTracker.ini
SD/IC705BT.ini
----------
/.backup.ini

Regarding the Touch not working, I can accept that it is a coincident. As i was running an application before without using Touch it could be that the M5Core2 was already broken before. No blame to you, but it could be that a bit or byte is fallen over in the Flash memory causing.

Happy with our developments and actually it started me using M5 products and i created a little application reusing code to control the IC705 for QO-100 GeoSatellite operation.

73 Johan, PA3ANG

closed 2022-10-26 by pa3ang