armel/M5HamRadio #16: Boot loop after installing M5HamRadioCore2

sixhills, 2023-03-31:
This may be a hardware fault, but I'd appreciate your kind advice.

I've purchased an M5Stack Core2 to run your brilliant Icom software with an IC-705.  I used M5Burner to install M5HamRadioCore2 and the burn appeared to work but the device then fails to boot.  The Serial Monitor reports a boot loop:

```Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060130  A0      : 0x80098eaf  A1      : 0x3ffbd920  
A2      : 0x3ffc6650  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x3ffbcee4  A7      : 0x80000001  A8      : 0x8010f729  A9      : 0x3ffbd900  
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x80000000  
A14     : 0x007bf4d8  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000 

Backtrace:0xfffffffd:0x3ffbd9200x40098eac:0x3ffbd940 

ELF file SHA256: 0000000000000000

Rebooting...
```
I can install UIFlow_Core2 using M5Burner and that works perfectly.  Some other firmwares burned using M5Burner also work but installing Core2FactoryTest also fails with a boot loop, this time reporting "Core  1 panic'ed (IllegalInstruction)".  This is repeatable: firmware that fails always fails; firmware that works always works.

Any advice is very welcome.

brianbgarber, 2023-03-31:
I wrote to M5Stack support about this very issue. Here is their reply:

"That should be the flash compatibility issue with old version IDF. We will update the firmware on M5Burner soon to fix it.

Pls wait for the news."

sixhills, 2023-03-31:
Thanks - that's most interesting.  I suppose we have to wait, unless there's some other method of installing the software.  I'm happy to install development environments, etc, if necessary.

armel, 2023-03-31:
Hello,

Thank you for your message. I'm sorry to read that you're experiencing this issue. I have never seen it before. To test and rule out the possibility of an error on my part, I just flashed a CORE2 and a CORE2 AWS, and everything is working normally. I will look into this on my end to see if I can find any additional information. However, this is the first time someone has been unable to flash their M5Stack using M5Burner. Really strange.

We keep in touch.

73'

Armel F4HWN.

brianbgarber, 2023-03-31:
Hi Armel, I don't think this is an issue on your software. I have just received my Core2 and have also had problems with the M5Burner software. I think M5Stack will have to update their code. Perhaps there is a slight modification to hardware?

armel, 2023-04-01:
Yes. It's possible for minor hardware modifications. I'm agree with you. It has happened in the past. For example, the IPS LCD panel has already evolved several times. The same goes for the USB-C connector (and the drivers...). It sometimes varies from one batch to another. But it still bothers me a little that some users encounter these problems, even if it is beyond my control. Hopefully, the M5Stack team will react quickly. In any case, thank you for your trust. We will stay in touch, and I leave this issue open.

73'

Armel F4HWN.

sixhills, 2023-04-01:
As an experiment, I've tried using esptool to install the software:

```esptool.py --port com[x] --baud 115200 write_flash --flash_mode dio 0x1000 ICMultiMeter.bin```

but that fails in the same way.

Is there any other method I could try?  If not, I'll wait and see if M5Stack issue a new version of M5Burner.  I'm sure that being able to use Armel's software will be worth the wait.

armel, 2023-04-01:
Hello,

Well no, this command will not work, for memory mapping reasons.
However, you can try this. 
In the packages/v0.0.5 directory of this repository, you will find a complete image for Core and Core2.

So, download the Core2_5a49628a4bf66e3f1afd9af73190946f.bin image (for your Core2).

Here is a direct link to download it, if necessary : https://github.com/armel/M5HamRadio/raw/main/packages/0.0.5/Core2_5a49628a4bf66e3f1afd9af73190946f.bin

This image contains the 3 apps ICSMeter, ICMultiMeter, and DXTracker in one huge firmware.
It is a complete raw 16MB dump.

You can now try to flash this image with this command :

`esptool.py --port com[x] --baud 921600 write_flash --flash_freq keep 0x000000 Core2_5a49628a4bf66e3f1afd9af73190946f.bin
`

Of course, change port and  baud rate (it works for me with 921600).

Keep me informed.

73'

Armel F4HWN.

sixhills, 2023-04-01:
Thank you so much for your helpful suggestions.  I've tried flashing the image as described but, unfortunately, it doesn't work.  After burning, the device enters an infinite boot loop with "Guru Meditation Error: Core  1 panic'ed (IllegalInstruction)."

I think there must be some subtle hardware change in new Core2 devices that make it incompatible with your software.  Even M5Stack's own factory test firmware fails.  It's very odd that some firmware programs work and some don't.

This is obviously an issue for M5Stack to fix.  It's not really your problem but I appreciate your time in trying to help.

armel, 2023-04-01:
Okay, thank you for your feedback. I'm sorry it doesn't work.
It's so strange. I hope M5Stack will find and communicate about this issue quickly.
It's the first time I heard about such a problem.
I'm hesitant to order a Core2 to try to understand what's going on.

73'

Armel F4HWN.

K5BRSTX, 2023-04-08:
Hello All.
I just received a new Core2 today.  I can load and run Core2FactoryTest and UNFlow by installing from the M5BUrner app.

armel, 2023-04-08:
Hello,

This is very strange. It's always the same error backtrace on M5Core2. Yet, I get many messages from users that it works without a problem. Maybe there is a problem on a recent batch of M5Core2?

About your M5Basic, is your SD formatted in FAT32? Can you show me the console trace? What is displayed on the screen? Can you make me a small video?

73' 

Armel F4HWN.

K5BRSTX, 2023-04-08:
Armel,
Thanks for the quick reply.

I do not know if this makes any difference, but when I run M5Burner I get a Java error.  I have attached a screenshot of that.  The program seems to run fine otherwise.
[image]

My M5Basic is a V2.6.  I purchased it last week and have not been able to get it to load M5HamRadio.  Could there be something in the V2.6 that has changed causing the issues?
Here is a screen capture of the start up and a screen shots of the program starting. My video was too big.
[image]
[image]
[image]
[image]
[image]

Here is a console screen shot of the startup.  I do not know if it will help any.
[M5Basic_Connect.txt](https://github.com/armel/M5HamRadio/files/11183430/M5Basic_Connect.txt)

I am going to check the M5 site for an updated burner app and also see if anyone else mentioned the Java issue.

Thank you so much for your assistance.
Benny K5BRS

K5BRSTX, 2023-04-08:
RE sdcard.  Yes it is a 32gb formatted as FAT32
[image]

I have tried a second 32gb and a 2gb (oldie) and get the same each time.

73's K5BRS

armel, 2023-04-08:
Ok, thank you. Well everything works fine on your M5Basic 😌 And I thought that the information message would be clear enough : _**Please, add a valid config file on LittleFS or SDCard**_. 🤗

The firmware can't find any valid configuration file, either in memory or on the SDCard. This is quite normal at this step.

I think you have missed an **important** step. It is up to you to create your configuration file with your settings which I cannot invent 🤔 I don't know which transceiver you will use, on which CI-V address, on which port (if you're using USB), etc. It's up to you to define it and make your own configuration file.

So please consult the documentation carefully. Prepare and format a micro SD card in FAT32. And follow these instructions : https://github.com/armel/M5HamRadio#icsmeter-icmultimeter-and-ickeyer-ini-file-settings

After that, if the micro SD card is correctly inserted, with some valid config files on it, you'll now see the Ini Loader with a blue gauge (step after Bin Loader...). You have 3 seconds to use the bottom middle button to select an .ini file that you put on it. Select the .ini file you would like to load and... enjoy.

73'

Armel F4HWN.

K5BRSTX, 2023-04-08:
I tried to follow exactly your instructions for the ini file.
Here is mine.  Did I miss something?
[IC705BT.txt](https://github.com/armel/M5HamRadio/files/11183520/IC705BT.txt)
I changed the extension so I could include it here

73' K5BRS

K5BRSTX, 2023-04-08:
I have two files on the 32gb sd.
DXTracker.ini  and IC705BT.ini

K5BRS

armel, 2023-04-08:
Ok, nice.
Your ini file seems good.
Do you see it appear in the list, at the Ini Loader stage? Did you select it ? Is it read without any error message on the screen ?

Armel F4HWN

K5BRSTX, 2023-04-08:
Armel,

Reference the 5 photos above.
In photo3 it is loading the ICSMeter.bin and completes at 100%.
In photo4 bin loader has popped up with a green bar.  I do not touch anything thinking that It should next show the ini loader stage.  
Photo 5 show what i get.  It does not show a ini loader stage.  Could there be something in the M5Basic V2.6 that is not seeing the sdcard slot?

I am thinking that I need to create a test so I can see if the sdcard slot is reading my card.

73 K5BRS

armel, 2023-04-08:
Ok, so according to your console, SD Cart is not mount.

```
[  1242][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
----------
[  5404][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work

```
The question is : why ?

A normal console message look like this...

Bin loader step...

```
19:53:54.238 > ----------
19:53:54.253 > SD/DXTracker.bin
19:53:54.297 > SD/ICKeyer.bin
19:53:54.311 > SD/ICMultiMeter.bin
19:53:54.324 > SD/ICSMeter.bin
19:53:54.337 > SD/MLX90640.bin
19:53:54.353 > SD/RRFRemote.bin
19:53:54.378 > ----------
```

Ini loader step ...

```
19:53:57.517 > ----------
19:53:57.528 > SD/Buggy.ini
19:53:57.535 > SD/DXTracker.ini
19:53:57.545 > SD/IC705BT.ini
19:53:57.555 > SD/IC705USB.ini
19:53:57.566 > SD/IC7300USB.ini
19:53:57.647 > SD/IC705BTK5.ini
19:53:57.655 > ----------
```

Finally, ini file load and parse...

```
19:54:00.791 > /.backup.ini
19:54:01.821 > YYYYY
19:54:01.821 > XXXXX
19:54:01.821 > 705
19:54:01.821 > A4
19:54:01.821 > 1
19:54:01.821 > 30 31 7D 33 B2 58 
19:54:01.821 > 115200
19:54:01.821 > http://192.168.1.32
19:54:01.827 > 1234
19:54:01.827 > 116000000.00
19:54:01.827 > 118000000.00
19:54:01.827 > 404000000.00
19:54:01.827 > 406000000.00
19:54:01.828 > ovf
```

Armel F4HWN.

brianbgarber, 2023-04-13:
Hello, M5Stack pushed out an updated Core2Factory test on the M5Burner and it now works for me.  I'm still having an issue burning the M5HamRadioCore2 v0.0.5. Here is what I caputred when trying to burn...

> Burning
> 
> --chip auto --port COM5 --baud 1500000 --before default_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x000 C:\Users\brian\Documents\M5Burner-v3-beta-win-x64\packages\firmware\5a49628a4bf66e3f1afd9af73190946f.bin
> esptool.py v4.1
> Serial port COM5
> Connecting...
> .
> .
> 
> Detecting chip type... Unsupported detection protocol, switching and trying again...
> Connecting...
> .
> 
> Detecting chip type...
>  ESP32
> Chip is ESP32-D0WDQ6-V3 (revision 3)
> Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
> Crystal is 40MHz
> MAC: 08:b6:1f:88:82:bc
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 1500000
> Changed.
> Configuring flash size...
> Auto-detected Flash size: 16MB
> Flash will be erased from 0x00000000 to 0x00ffffff...
> Compressed 16777216 bytes to 6328657...

armel, 2023-04-13:
Hi,

Oh thank you. 
I have the same thing, but the flashing process goes to the end 🤔
Here is the complet trace what I get from my side.

```
Burning
--chip auto --port /dev/tty.usbserial-02041F5C --baud 1500000 --before default_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x000 /Applications/M5Burner.app/Contents/Resources/packages/firmware/5a49628a4bf66e3f1afd9af73190946f.bin
esptool.py v3.3.2-dev
Serial port /dev/tty.usbserial-02041F5C
Connecting...
.
.

Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
.
.
.
.

Detecting chip type...
 ESP32
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b8:f0:09:c4:71:20
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00000000 to 0x00ffffff...
Compressed 16777216 bytes to 6328657...
Writing at 0x00000000... (0 %)
Writing at 0x0001102a... (0 %)
Writing at 0x00017d87... (0 %)
Writing at 0x0001fca2... (1 %)
Writing at 0x000274e3... (1 %)
Writing at 0x000310d6... (1 %)
Writing at 0x00037227... (1 %)
Writing at 0x0003cb71... (2 %)
Writing at 0x000426a5... (2 %)
Writing at 0x00048837... (2 %)
Writing at 0x0004e8ca... (2 %)
Writing at 0x00054651... (3 %)
Writing at 0x0005a75a... (3 %)
Writing at 0x000603b0... (3 %)
Writing at 0x000663c3... (3 %)
Writing at 0x000704d9... (4 %)
Writing at 0x0007d004... (4 %)
Writing at 0x00086970... (4 %)
Writing at 0x00090969... (4 %)
Writing at 0x000a091a... (5 %)
Writing at 0x000a64b8... (5 %)
Writing at 0x000ac1a6... (5 %)
Writing at 0x000b1b6d... (5 %)
Writing at 0x000b6ed0... (6 %)
Writing at 0x000bbf2d... (6 %)
Writing at 0x000c0f4c... (6 %)
Writing at 0x000c6827... (6 %)
Writing at 0x000cc5f9... (7 %)
Writing at 0x000d24bb... (7 %)
Writing at 0x000d810e... (7 %)
Writing at 0x000ddadb... (8 %)
Writing at 0x000e3985... (8 %)
Writing at 0x000e934a... (8 %)
[... 339 lines cut ...]
Writing at 0x00f8e9dd... (96 %)
Writing at 0x00f9482a... (96 %)
Writing at 0x00f9ae4c... (96 %)
Writing at 0x00fa7896... (97 %)
Writing at 0x00fb3a08... (97 %)
Writing at 0x00fbe805... (97 %)
Writing at 0x00fcbbd0... (97 %)
Writing at 0x00fd80d9... (98 %)
Writing at 0x00fddcb4... (98 %)
Writing at 0x00fe3e14... (98 %)
Writing at 0x00fe8f85... (98 %)
Writing at 0x00fee1d2... (99 %)
Writing at 0x00ff30d6... (99 %)
Writing at 0x00ff87e2... (99 %)
Writing at 0x00ffe501... (100 %)
Wrote 16777216 bytes (6328657 compressed) at 0x00000000 in 106.2 seconds (effective 1263.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

And it works.

Armel F4HWN.

brianbgarber, 2023-04-13:
Ah, gotcha. I tried to burn version 0.0.5 again, and here is the output from the COM monitor

> [FROM M5Burner] COM5 opened.
> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
> Memory dump at 0x4010f730: 16ee8366 9e810054 0008e00d
> Core  1 register dump:
> PC      : 0x4010f734  PS      : 0x00060130  A0      : 0x80098eaf  A1      : 0x3ffbd920  
> A2      : 0x3ffc6650  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
> A6      : 0x3ffbcee4  A7      : 0x80000001  A8      : 0x8010f73c  A9      : 0x3ffbd900  
> A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00000000  
> A14     : 0x007bf4d8  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> Backtrace:0x4010f731:0x3ffbd9200x40098eac:0x3ffbd940 
> 
> 
> ELF file SHA256: 0000000000000000
> 
> Rebooting...
> ets Jul 29 2019 12:21:46
> 
> rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1184
> load:0x40078000,len:12812
> load:0x40080400,len:3032
> entry 0x400805e4
> Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
> Memory dump at 0x4010f730: 16ee8366 9e810054 0008e00d
> Core  1 register dump:
> PC      : 0x4010f734  PS      : 0x00060130  A0      : 0x80098eaf  A1      : 0x3ffbd920  
> A2      : 0x3ffc6650  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
> A6      : 0x3ffbcee4  A7      : 0x80000001  A8      : 0x8010f73c  A9      : 0x3ffbd900  
> A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x00000000  
> A14     : 0x007bf4d8  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000000  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> 
> Backtrace:0x4010f731:0x3ffbd9200x40098eac:0x3ffbd940 
> 
> ELF file SHA256: 0000000000000000
> 

and then repeats.  

Thanks!

armel, 2023-04-13:
Hello,

Ok, thank you.
I added a comment on this open issue (with our console trace).

https://github.com/m5stack/M5Core2/issues/127#issuecomment-1507211498

And I contacted the M5Stack community manager.

Stay tuned 🤔

Armel F4HWN.

sixhills, 2023-04-13:
The new version (v2.1) of Core2FactoryTest works for me, too (whereas the previous version v1.3.2 failed).  But, as expected, burning v0.0.5 of M5HamRadioCore2 still fails.  Mine fails with InstrFetchProhibited:

```
ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:12812
load:0x40080400,len:3032
entry 0x400805e4
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060130  A0      : 0x80098eaf  A1      : 0x3ffbd920  
A2      : 0x3ffc6650  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x3ffbcee4  A7      : 0x80000001  A8      : 0x8010f729  A9      : 0x3ffbd900  
A10     : 0x00000000  A11     : 0x00060023  A12     : 0x00060023  A13     : 0x80000000  
A14     : 0x007bf4d8  A15     : 0x003fffff  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace:0xfffffffd:0x3ffbd9200x40098eac:0x3ffbd940 
```

So M5Stack have clearly fixed the fault with burning the factory test firmware to recent Core2 devices and this implies that there's a change that Armel could make to M5HamRadioCore2 that would make it compatible with our recent Core2s.  The mystery is what that change should be and why it's necessary.

I logged that Github issue 3 days ago and it's disappointing that M5stack have fixed the compatibility issue with the factory test firmware without making any response at all on the open Github issue.

By the way, it's not just your firmware, Armel, that's affected by this issue. I've tried burning most of the programs listed in M5Burner and about half of them work and the other half fails.

Your program has had far more downloads than anyone else so it's in M5stack's interest to fix this.  I think that your program has generated many sales of the Core devices.

armel, 2023-04-15:
Hi Guys,

I don't forget you. But it is difficult for me to understand the origin of the problem, since I can't reproduce it. 

So I am exploring several possibilities. I have prepared a special firmware for the Core2 (touch screen). You can find it, on M5Burner, under the name "F4HWN CORE2 TEST" (remember to refresh the M5Burner firmware list). It only embeds the ICSMeter.

Does it work on your computer? Please let me know. 

Armel F4HWN.

sixhills, 2023-04-15:
It works!  Thank you so much for persevering with this issue.

May I ask what change you made to the software?

armel, 2023-04-15:
Okay, thank you for your response, that's good news. 

But there's going to be some bad news. Apparently, recent Core2s are a bit different and require the use of the latest version of the ESP-IDF SDK developed by Espressif. I don't know why. 

The problem is that this SDK has evolved a lot. It is much larger, and this directly affects the size of my firmware and the memory partition schema. And I am no longer able to provide a large firmware with ICSMeter, ICMultiMeter, and DXTracker.

I will have to publish firmware for each application on M5Burner. And It will no longer be possible to switch from one firmware to another, even using an SD card. Each time, M5Burner (or esptools via CLI) will have to be used. And that's really too bad. It means that if you want ICSMeter and ICMultiMeter, the simplest solution will be to have 2 M5Stacks. Whereas before, you could restart and switch from one application to another via BinLoader.

By the way, which applications are you most interested in? I will try to package them and put them online cleanly during this weekend.

Well, maybe I'll find another solution over time. But for now, that's where we are. I'm really doing my best, but I'm not a wizzard.

Thank's again for your tests and reports.

Armel F4HWN.

brianbgarber, 2023-04-15:
I also wanted to report that I got the CORE2 TEST to work. 

It's a shame about the changes to the SDK (and certain bad players that take advantage of open software) :-( but as you mentioned maybe something will change in the future to make it possible to load all three at once. 

Thank you for your continued hard work on this project!

sixhills, 2023-04-15:
Thanks for the explanation, Armel.  I'm sorry you're having all this trouble, entirely due to M5Stack making an undocumented hardware change and not testing against their own published test software. 

I  should say that the new version of ICSMeter boots OK and the Settings menu works but i've not yet been able to pair it with my IC-705.  The reason seems to be that the Core2 isn't detecting the SD card, which is the same problem that @K5BRSTX mentioned above with his M5 Basic.   Perhaps this is another hardware change?

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
[  1742][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
----------
[  2751][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
Loading config file...
/.backup.ini
Find ini file.
Ini file is valid.
F1ZPX
petitchaton
705
A4
1
30 31 7D 33 B2 58 
115200
http://192.168.1.32
1234
116000000.00
118000000.00
404000000.00
406000000.00
ovf
Ini file parsing is finished.
Wifi Client Disconnected
Wifi Client Disconnected
Wifi Client Disconnected
Wifi Client Disconnected
[ 14507][E][BluetoothSerial.cpp:378] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT failed!, status:1
Wifi Client Disconnected
Wifi Client Disconnected
Wifi Client Disconnected
Attempt 1 - Make sure IC-705 is available and in range.Wifi Client Disconnected
Wifi Client Disconnected
Wifi Client Disconnected
[ 24487][E][BluetoothSerial.cpp:378] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT failed!, status:1
Wifi Client Disconnected
Wifi Client Disconnected
Attempt 2 - Make sure IC-705 is available and in range.Wifi Client Disconnected
```
73, Mike G4GOC.

sixhills, 2023-04-15:
I've re-installed Core2FactoryTest and that also fails to detect the SD card.  The display reports " SDcard Find Failed" and the serial console reports "[  2053][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work".  I'll try a different SD card later but this looks like a serious (hardware?) problem.

armel, 2023-04-15:
Hummm, first, could you test the new firmware I just uploaded.

You can find it, on M5Burner, under the name "F4HWN CORE2 TEST 2" (remember to refresh the M5Burner firmware list). I tried to optimize the memory... A big challenge. This firmware embeds the very last version of ICSMeter, ICMultiMeter and DXTracker. So I hope it will make possible, again, to change the firmware via the BinLoader.

Now, about SD mount failed, well, this is a new huge problem !!! And if the official firmware bugs too, it's not a good news.

Well, let's proceed step by step. Can you tell me if this new firmware starts and if you can use the binloader by loading another firmware in memory. Thank you for your help. 

Armel F4HWN.

sixhills, 2023-04-15:
Apologies - the failure to detect the SD card was caused by me using a 32GB card.  When I replaced it with a 16GB card, Core2FactoryTest detects the card successfully.  So there's not a new problem!

I've installed v0.0.2 of your "F4HWN CORE2 TEST 2" firmware.  It installs OK, reads my SD card successfully and pairs to my IC-705!  I've tested ICSMeter, ICMultiMeter and DXTracker and they all work perfectly.

So you've completely fixed the compatibility issue with new Core2s.  I'm very impresed (you are a wizard!) and very pleased with the result.  Thank you so much.

I'll leave the issue open for a while so that the others can confirm that it works for them.

armel, 2023-04-15:
Oh yes, so we finally got it 🤗

That's great news. I'm so relieved. It was not easy to solve, but you all helped me a lot. The quality of your reports has been very helpful. Failure was not an option ! 

By the way, Mike, I am very honored by your gesture. It is very kind of you. Thank you very much and 73 😊

Armel F4HWN

closed 2023-04-17 by sixhills