armel/M5HamRadio #33: M5 Core 2 Touch Screen only - ini file will not load

fbeider, 2023-05-05:
Hello

I have a M5 core2 touch screen only (no physical buttons).  

Using M5 burner I installed .8 for the core 2.

Get the error message Ini file does not exist
failed to load config file

Here is the ouptut  of the serial monitor.  The .ini file appears to be loading from the SD card

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
----------
----------
----------
SD/IC705BT.ini
----------
/.backup.ini
[FROM M5Burner] Serial Port Disconnected.

armel, 2023-05-05:
Hello,

Take a look to the README again : https://github.com/armel/M5HamRadio#ini-loader
You need to select your ini config file during the Ini Loader step (blue progress bar). 

Armel F4HWN.

fbeider, 2023-05-05:
I cannot select a config file as there are no hardware buttons on M5 Core 2 touch screen only.   No buttons appear on the screen to touch.  I do see the inil oader v.07 on the screen

armel, 2023-05-05:
Sure, of course you've got (virtual) buttons. Take a look a this image...

[image]

Armel F4HWN.

armel, 2023-05-05:
Ok, so during the Ini Loader step (blue progress bar), 

1 - touch middle virtual button to enter into Ini Loader
2 - use left / right virtual buttons to choose your ini file.
3 - touch middle virtual button to select it.

Armel F4HWN.

fbeider, 2023-05-06:
I got it working.

Thank You!

armel, 2023-05-06:
Good news.
Thank you.
I close this issue.

73'

Armel F4HWN

closed 2023-05-06 by armel