armel/M5HamRadio #39: ini files not read, SD card not detected on M5Stack Basic using 0.0.8

ZigmundRat, 2023-05-16:
I am at a loss. Hopefully you can provide some debug help here. 

I've done everything I can think of to solve this. The ini files have been customized and are not at default. I have used Notepad in Windows, as well as Visual Studio Code to create them and have used UTF-8 and ANSI formats. CRLF or LF only line ending. I have used 2/4/8GB SD cards, HC class4, class 10, all Sandisk with no difference. The cards have been formatted FAT32 with a single partition. When M5HamRadio runs, I never see the ini loader appear, and I never see any lines in the log that show it even reading the card. There are no lines with /SD/... showing that it's read the card. I only get the error message about placing a valid config file in LilttleFS (which I don't know how to do) or the SD card (which isn't read). I also tried the SD-Updater application - no problem reading and loading apps from the same cards and partitions. Same for the default demo shipped with the M5 Basic. It can see and read/write the card no problem (partial output below):

Listing directory: /
  DIR : /System Volume Information
  FILE: /ICKeyer.bin  SIZE: 2107936
  FILE: /ICMultiMeter.bin  SIZE: 2123904
  FILE: /ICSMeter.bin  SIZE: 2214816
  FILE: /DXTracker.bin  SIZE: 1523920
  FILE: /DXTracker.ini  SIZE: 146
  FILE: /IC7300USB.ini  SIZE: 548
Writing file: /hello.txt
File written
Reading file: /hello.txt
Read from file: Hello world
---
Contents of IC7300USB.ini:
; Icom Config
[icom]
icom_model = 7300
icom_address = 0x94
icom_connect = USB
icom_device = /dev/ttyUSB0

; Wifi Config
[wifi]
wifi_ssid = xxx
wifi_password = xxxxxxxxx

; Proxy Config
[proxy]
baud_rate = 115200
proxy_url = http://192.168.1.32
proxy_port = 1234

; Transverter Config
[transverter]
transverter_lo_1 = 116000000
transverter_lo_2 = 118000000
transverter_lo_3 = 404000000
transverter_lo_4 = 406000000
transverter_lo_5 = 9968000000

; Geolocation Config
[geolocation]
latitude = xx.2872
longitude = -yy.5333

Contents of DXTracker.ini:
; Wifi Config
[wifi]
wifi_ssid = xxx
wifi_password = xxxxxxxxx

; Geolocation Config
[geolocation]
latitude = xx.2872
longitude = -yy.5333
---
Console log from M5HamRadio:
[FROM M5Burner] COM3 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (462) psram: PSRAM ID read error: 0xffffffff
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
----------
Loading config file...
/.backup.ini
Ini file does not exist.

armel, 2023-05-16:
Hi,

According to the log, it's like your SD card is not correctly inserted or mounted.
Here is a log when SD card IS NOT inserted (it looks like yours) :

```
[FROM M5Burner] COM4 opened.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (462) psram: PSRAM ID read error: 0xffffffff
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
----------
Loading config file...
/.backup.ini
Ini file does not exist.
```

And here is a log when SD card IS inserted (same M5Stack...) :

```
[FROM M5Burner] COM4 opened.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (462) psram: PSRAM ID read error: 0xffffffff
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
SD/DXTracker.bin
SD/ICKeyer.bin
----------
----------
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
/.backup.ini
```

Not that, according to this log, there are 2 binaries on SD Card (SD/DXTracker.bin and SD/ICKeyer.bin) and 5 ini config files (SD/IC7300USB.ini, SD/Buggy.ini, SD/DXTracker.ini, SD/IC705BT.ini and  SD/IC705USB.ini).

I doubt that it's a software issue. Have you try to clean the contacts on the micro SD card ? Every time users had problems, it was due to the SD card, bad contacts or dirty contacts.

73'

Armel F4HWN

ZigmundRat, 2023-05-16:
I read about the SD card issues in the documentation. And reread everything I could, including the other closed issue. I tried everything I could before opening this report. 

I just now replaced the M5HamRadio app with the factory one, *without touching the SD card*, rebooted, and got the same output showing the contents of the SD card:

Listing directory: /
DIR : /System Volume Information
FILE: /ICKeyer.bin SIZE: 2107936
FILE: /ICMultiMeter.bin SIZE: 2123904
FILE: /ICSMeter.bin SIZE: 2214816
FILE: /DXTracker.bin SIZE: 1523920
FILE: /DXTracker.ini SIZE: 146
FILE: /IC7300USB.ini SIZE: 548
Writing file: /hello.txt
File written
Reading file: /hello.txt
Read from file: Hello world

Could there be some hardware revision that is affecting the library you are using for SD access?

armel, 2023-05-16:
Ok. What factory application are you talking about?

About an hardware revision affecting the library I use for SD access, I don't think so. Or maybe I'm just not aware of it. But I'm using recent and up to date versions. So... it's really strange.

Armel F4HWN.

ZigmundRat, 2023-05-16:
I admit that it's strange. The 'factory application' is just the demo that ships with it. It does some graphic demos, scans the wifi, lists the TF card contents, writes and reads back a test file on the SD card, and finally goes to a button test. I can send it to you if you want. 

I found more SD cards: Emtec, Verbatim, PNY, Micro Center (computer store generic card) , and a Kingston. I will do the same test with all of them and let you know the outcome. 

Are you using PlatformIO or Arduino for your builds? Maybe I'll do some independent testing.

armel, 2023-05-16:
Ok thanks. The source code of the factory application you're using is probably this one : 
https://github.com/m5stack/M5Stack/blob/master/examples/Basics/FactoryTest/FactoryTest.ino

I use different (and newer) libraries. And yes, that's right, I'm using PlatformIO to code. 

Your M5Stack BASIC is a V2.7 model (a very recent model) ?

Armel F4HWN.

ZigmundRat, 2023-05-17:
Yes, that look like what came loaded on them (I have two). I see a revision number on the PCB silkscreen - it says v2.61. 

If you can share which library you are using I should be able to create a simple 'sketch' to try it on my hardware. If it's not public, well then maybe you can give me a debug version that does more logging. I have a mental list of things it could be, but that's probably not very helpful. 

Paul - K9MR

armel, 2023-05-17:
Okay, Paul. Thanks for all these informations. 

There is good news and bad news. 

The good news is that I'm sure we can solve the problem, even if I don't know the exact cause yet. But failure is not an option ;) We will fix it! 

The bad news is that you will probably have to be patient. I'm currently running out of time because I have a lot of work to do. I think the problem is in a low-level library (which I am not the author of). 

I will do my best to write a small test code in the next days. But I can't promise anything, and it may take until the end of the month for me to have time to look into the problem. 

I keep this thread open and keep you informed. 

Armel F4HWN.

armel, 2023-05-17:
Hi Paul,

Thank you for your help. I've just uploaded a small debug firmware, available on M5Burner. It's called "**SD Card Test**". When you have the time, could you flash your M5Stack BASIC with this firmware and let me know what you see in the console (Serial Monitor)? Of course, remember to insert an SD card into your M5Stack (with some .ini and / or .bin files on it).

73'

Armel F4HWN.

ZigmundRat, 2023-05-18:
Hi Armel -
Here's the results:

[FROM M5Burner] COM3 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (125) psram: PSRAM ID read error: 0xffffffff
Test 80Mhz
Test 40Mhz
Test 25Mhz
Test 20Mhz
Test 15Mhz
SD/ICKeyer.bin
SD/ICMultiMeter.bin
SD/ICSMeter.bin
SD/DXTracker.bin
_____
SD/IC7300USB.ini
SD/DXTracker.ini
_____
Test 10Mhz
SD/ICKeyer.bin
SD/ICMultiMeter.bin
SD/ICSMeter.bin
SD/DXTracker.bin
_____
SD/IC7300USB.ini
SD/DXTracker.ini
_____

Kind of funny, because I was also thinking about the SPI clock. Isn't the default 20MHz? I wonder if this means the pullups are not correct. I don't know if there are internal or only external on the ESP32 or on this device. 

Thanks for the quick work :)

Paul - K9MR

armel, 2023-05-18:
Oh, thank you Paul :) Very interesting. It confirms my hypothesis. So, here we go, we have identified the source of the problem. Remember, failure is not an option :)

I've tested it on 2 differents M5Stacks, chosen randomly. And here's what I obtained with them:

```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (125) psram: PSRAM ID read error: 0xffffffff
Test 80Mhz
SD/menu.bin
SD/DXTracker.bin
SD/ICKeyer.bin
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
Test 40Mhz
SD/menu.bin
SD/DXTracker.bin
SD/ICKeyer.bin
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
Test 25Mhz
SD/menu.bin
SD/DXTracker.bin
SD/ICKeyer.bin
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
Test 20Mhz
SD/menu.bin
SD/DXTracker.bin
SD/ICKeyer.bin
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
Test 15Mhz
SD/menu.bin
SD/DXTracker.bin
SD/ICKeyer.bin
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
Test 10Mhz
SD/menu.bin
SD/DXTracker.bin
SD/ICKeyer.bin
----------
SD/IC7300USB.ini
SD/Buggy.ini
SD/DXTracker.ini
SD/IC705BT.ini
SD/IC705USB.ini
----------
```

As you can see, they pass all the tests successfully, regardless of the clock frequency. It's incredible. In fact, we have to face the fact that there are some subtle differences from one M5Stack production batch to another. I'm not sure of the exact reason of this. I also have the impression that it depends on the class of the SD card.

I think I'll fix the frequency at 10 MHz in the next release, just to be safe.

Now, I need to package a new version. However, as I mentioned, I'm not sure if I'll be able to do it quickly. Currently, I only have a basic and poor Windows PC with me (I hate it...). Also, I'd like to take the time to run another round of tests with other M5Stacks. Can you wait for a few more days. Last but not least, I am also waiting for the last new M5Stack CORE2 S3 that I should receive at the end of the next week. So could you wait about 10 days ? I know it's long, but I try to do my best with the best quality service. 

Thank you again for your help.

73'

Armel F4HWN.

ZigmundRat, 2023-05-18:
Some more info for you:

I tested 16 cards. Most are Sandisk, but also Emtec, Verbatim, Kingston, PNY, MicroCenter, and unmarked generic. The ONLY ones that worked from 80 on down are the one generic, and one Verbatim that is not otherwise marked. What's funny is I avoided them because they were not marked.

EVERY Sandisk card only works from 15 and below. EVERY card marked HC Class 4 or Class 10 only works from 15 and below. AFAIK Class 10 is supposed to support 50, and every card is supposed to support 25. 

Unfortunately I don't have another platform to test these cards in so I don't know if it's the cards or the M5Stack. If I get curious enough I'll put the cope on it and see what the waveforms look like. I suppose besides a poor implementation on these v2.61 M5Stack units, they could be counterfeit, or the supposed Sandisk cards could be (but they were all reliably sourced). 

Since I have two cards that work at 80, I will try them with the standard application. Then we will see how much time difference there is loading using the lower rate and the other cards. Maybe you can do the card detection at increasing rates until it fails and use the last successful one. I don't know if you want to penalize everyone for someone's (my?) slow interface. 

Thanks so much for working through this. We did identify a definite problem, and have a fix. Well done!

armel, 2023-05-18:
Oh, that's very interesting. Thank you for all these tests. The SD card I used is a Gigastone, 32 GB, Class 10. I know I have no issue with Netac too (an unknown brand to me).

I agree that implementing a procedure to detect the optimal speed would be the best solution. You're right. I will try to code all this in the next release.

Paul, if you have time to see what the waveforms look like with your scope, I'm interested too. I think that the components used in the M5Stack are not necessarily of extraordinary quality. But an M5Stack costs less than 50 USD... It remains a great platform and I love it :) My first Basic is nearly 3 years old and still works perfectly.

73,

Armel F4HWN.

armel, 2023-05-27:
Hi Paul,

I have just published a new version 0.0.9 of the M5HamRadio.
This version mainly improves micro SD card detection and should, I hope, solve the problem encountered with some micro SD cards. There's an initial check at 40MHz and if that doesn't work, I downgrade to 10MHz.
I hope this will fix this issue :) 

73'

Armel F4HWN.

ZigmundRat, 2023-06-04:
Finally had a chance to load this and test. First, the bin and ini loader versions reported are still 0.8 - I'm not sure if that was an oversight or what. 

If I let the unit boot by itself, it goes through the 'bin loader' and 'ini loader' complains about 'ini file does not exist. Failed to load config file.'  (Maybe you could add the name to the error message? I don't know what ini file it's looking for. )  BTW, the documentation on the ini file usage is a bit unclear to me - I don't know which file names it looks for when. That's why you see the different ones there on the card. I also don't know if it needs CRLF, or just LF as line ending characters, or if it matters, or if any of it is case sensitive.

But the console output is interesting. Notice the sequence of attempts. (Please see below.) 

If I interrupt the bin loader and select DXTracker.gz, it loads, then invokes 'bin loader' again followed by 'ini loader' and again complains about 'ini file does not exist'.

Still using DXTracker as an example, if I interrupt the ini loader, it shows the ini files under 'SD Card Storage'. I select DXTracker.ini and it next says 'ini file invalid', then 'ini file parsing is finished', and then transitions to the screen with QR code. (I may have missed a message, but that's what I can see.) The thing is, it appears to load at least some parameters. With DXTracker for example, it does connect to WiFi and retrieve data - but I don't know how to tell if the lat/lon is read correctly. . The same 'ini file invalid' happens for the other programs and associated ini files as well.

Aftrer doing all of the above, if I remove the SD card and reboot, it loads DXTracker and good WiFi parameters and does not complain. 

So I think there has been some progress, but something is still not right. 

Conosole output:
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (329) psram: PSRAM ID read error: 0xffffffff
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
-
Test 40 Mhz
SD/DXTracker.ini
SD/IC7300USB.ini
SD/ICMultiMeter.ini
SD/ICSMeter.ini
SD/ICKeyer.ini
----------
File index 5
/.backup.ini

DXTracker.ini file:

; Wifi Config
[wifi]
wifi_ssid = ssid
wifi_password = password

; Geolocation Config
[geolocation]
latitude = X3.2872
longitude = -Y2.5333
```

armel, 2023-06-04:
Hi Paul,

Thank you for your message.

First here are the versions changes :

- M5HamRadio was in version 0.0.8 and move to 0.0.9
- BinLoder and IniLoader **is a sort of projet in the project** and were in version 0.0.7 and move to 0.0.8
- Each version of applications changed too.

On first restart, what you see is normal. The IniLoader is not looking for any particular file. **There is no default file name to load or whatever.** It's up to you to tell it which one to choose (at least at the first time). By the way, IniLoader will complain about "ini file does not exist. Failed to load config file" until you load a valid config file (and even if you change binary as you did when loading the DXTracker.gz).

After that, according to the README

When you select an .ini file with the Ini Loader (and if it is valid), a copy is made on the Flash memory of the M5Stack. And if at the next reboot you do not make any action / selection of .ini file with the Ini Loader, it is this last valid .ini file, in Flash memory, that will be read automatically. This way, you can remove the SD card from your M5Stack, if you wish.

So, if you change something in your .ini file (for example, the Wifi password, etc.), you must restart your M5Stack and select it again from the Ini Loader to force reload this new version. A new copy will then be made in the M5Stack Flash memory. And it will be this new version that will be read at the next reboot (if no action / selection with the Ini Loader).

In fact, IniLoader do a copy of the last valid .ini file on the LittlesFS flash memory and rename it .backup.ini. 

And yes, the .ini file contend is case-sensitive. About line ending, it don't care.

About latitude and longitude, you need to use decimal coordinates. So without letters... It's maybe why you see the ini file invalid message.

So for example :

- Paris (France) : latitude 48.692660, longitude 2.424202
- Los Angeles (USA) : latitude 33.876223, longitude -118.125558
- Sydney (Australia) : latitude -33.734333, longitude 151.054459

73'

Armel F4HWN

ZigmundRat, 2023-06-04:
Fair enough. This problem (reading from SD with lower speed cards) does seem to be fixed. And it all seems to work.

There are a few things that remain which were (are?) slightly confusing.
1) Loading a bin file using bin loader: The bin file is loaded, then bin loader is immediately executed again. This made me think that the bin file was not loaded the first time. I expected it to just proceed to ini loader. I guess what happens is that after loading the bin file, it resets - which makes it run the bin loader again. So this is in fact normal behavior.

2) Selecting ANY ini file from SD card using ini loader results in the 'ini file is invalid' message, even if it later appears to work.

3) Even if a good ini file has been saved to internal flash, ini loader will ALWAYS say invalid ini file if there is a SD card inserted.

My actual DXTracker ini file is below.  This is copied from the github with the ssid/password and lat/lon changed for my location. Either it's invalid in some way (please explain) or the ini loader is producing misleading messages. I do not get any messages that indicate it could not parse the contents. 

```
; Wifi Config
[wifi]
wifi_ssid = IOT
wifi_password = IOTIOTIOT

; Geolocation Config
[geolocation]
latitude = 43.2872
longitude = -72.5333
``` 

Thanks for the support!
Paul - K9MR

armel, 2023-06-04:
Hi Paul,

1) Yes, that's how it works. The binLoader is executed at every startup. Like the iniLoader, it's part of the standard startup procedure for my applications, offering the possibility of starting another one if necessary. By the way, in my humble opinion, there's no other way to perform this magic trick, which allows several applications to coexist on the same M5Stack.  

2 and 3) I tried your ini file (I just modify it for a valid ssid and password). It works and I see these messages :

- Find ini file.
- **Ini file is valid.**
- Ini file parsing is finished.

And the application start. So I can't reproduce what you see.

Here is a short video : https://www.youtube.com/shorts/EZiykl4QvOA

And here is the console log :

```
21:03:29.349 > ets Jun  8 2016 00:22:57
21:03:29.350 > 
21:03:29.350 > rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
21:03:29.354 > configsip: 0, SPIWP:0xee
21:03:29.354 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:03:29.360 > mode:DIO, clock div:1
21:03:29.365 > load:0x3fff0030,len:1344
21:03:29.365 > load:0x40078000,len:13924
21:03:29.371 > ho 0 tail 12 room 4
21:03:29.371 > load:0x40080400,len:3600
21:03:29.371 > entry 0x400805f0
21:03:29.841 > E (329) psram: PSRAM ID read error: 0xffffffff
21:03:30.104 > ----------
21:03:30.138 > LittleFS/DXTracker.gz
21:03:30.153 > LittleFS/ICKeyer.gz
21:03:30.169 > LittleFS/ICMultiMeter.gz
21:03:30.185 > LittleFS/ICSMeter.gz
21:03:30.185 > ----------
21:03:30.215 > Test 40 Mhz
21:03:30.224 > SD/DXTracker.bin
21:03:30.239 > SD/RRFRemote.bin
21:03:30.243 > SD/M5HamRadio.bin
21:03:30.247 > SD/ICKeyer.bin
21:03:30.253 > SD/ICMultiMeter.bin
21:03:30.258 > SD/ICSMeter.bin
21:03:30.265 > SD/M5Unit-Thermal1-AF.bin
21:03:30.272 > SD/M5Unit-Thermal2-AF.bin
21:03:30.302 > ----------
21:03:30.384 > ----------
21:03:30.384 > File index 12
21:03:32.557 > ----------
21:03:32.558 > SD ready at 40 Mhz
21:03:32.567 > SD/Bug.ini
21:03:32.567 > SD/IC705BT.ini
21:03:32.573 > SD/IC705USB.ini
21:03:32.576 > SD/IC7300USB.ini
21:03:32.643 > ----------
21:03:34.854 > 00 04 00
21:03:39.472 > /Bug.ini
21:03:40.483 > MySSID
21:03:40.484 > MyPassword
21:03:40.484 > 43.29
21:03:40.485 > -72.53
21:03:41.499 > Wifi Client Connected
21:03:42.129 > 192.168.1.47
```

73'

Armel F4HWN.

ZigmundRat, 2023-06-04:
Thanks Armel - Everything is good now. I apologize for the trouble. I recreated the file from scratch and it all works OK now. I don't know what the issue was :/ Maybe the card is still 'iffy'? Thank you so much for the fix and the excellent support.

Best 73s,
Paul - K9MR

armel, 2023-06-04:
Ok Paul. Thank you too, for this interesting exchange.
My pleasure 🤗

73'

Armel F4HWN.

closed 2023-06-06 by armel