armel/M5HamRadio #49: Documentation improvement

cdaller, 2023-07-29:
Hello Armel,

thanks for this great piece of software! Really appreciate you work!

I managed to connect my M5Stack fire to my IC705 via bluetooth after having set the following parameters - maybe you could add  IC705 BT setup instructions):

* Menu > Set > Connectors > CI-V > CI-V USB Echo Back OFF
* Menu > Set > Bluetooth Set > Data Device Set > Serialport Function CI-V (Echo Back  OFF)
* ini file: set icom_address to address from Menu > Set > Connectors > CI-V > CI-V Address

Cheers,
Christof

armel, 2023-07-30:
Hi Cristof,

Thank you for your message. It's a pleasure.
Ok, I could try to improve the IC705 BT setup instructions. 

About **Menu > Set > Bluetooth Set > Data Device Set > Serialport Function CI-V (Echo Back OFF)**, are you sure ? Mine is set as **Echo Back ON** and it works. 

Regards,

Armel.

cdaller, 2023-07-31:
Hi Armel,

Hmmm - it works with "Echo Back OFF" - have not tried it otherwise. If you try and it works, it will probably work with me as well (cannot try it until next weekend). So maybe this is not important to set.

regards,
Christof

closed 2023-10-28 by armel