armel/M5HamRadio #55: ini file error

JohnShep12, 2023-08-25:
Hi I am having issues trying to finish of the install on my M5stack core 2, I get an error with ini file does not exist.  when I try a IC705bt.ini file I have modified from your information, I have also tried just installing the default example but get an error Ini file parsing error.

armel, 2023-08-25:
Hi,

Please, send me your ini file.
I will take a look on it.

Regards,

Armel F4HWN.

JohnShep12, 2023-08-25:
Thanks Armel.

73’s John

> 
> 
> Hi,
> 
> Please, send me your ini file.
> I will take a look on it.
> 
> Regards,
> 
> Armel F4HWN.

John Sheppard
***@***.***

￼

armel, 2023-08-25:
My personal email address is on my QRZ.com page 
https://www.qrz.com/db/F4HWN

73'

Armel F4HWN.

closed 2023-09-23 by JohnShep12