armel/M5HamRadio #60: Invalid INI Files

TommyMartin, 2023-10-28:
I have been trying to get these M5HamRadio applications to work with my IC705. I get errors when the ini is loading saying that it is invalid.

The error comes after the firmware is loaded on the device and the ini loader is attempting to read the ini. There is a brief message saying the ini is invalid. The app loads after the message but doesn't seem to have my settings.

I have tried a modified one with my information as well as the one checked into the repository by the name of ic705bt.ini

Any help is appreciated.

TommyMartin, 2023-10-28:
All that said.... I did use notepad for my files and I copied the raw INI from github so there should be no embedded characters in the files.

armel, 2023-10-28:
Could you send me your ini file ?

TommyMartin, 2023-10-28:
Sure thing. Thank you for the quick response. 

I had to rename the extension to send them. I get the same message with both.

I have the M4Stack Core2 AWS one. Since i sent this I have managed to pair to my 705 but I still get the message. I am wondering if it is related to my GPS coordinates.

[dxtracker.ini.txt](https://github.com/armel/M5HamRadio/files/13196264/dxtracker.ini.txt)
[ic705zno.ini.txt](https://github.com/armel/M5HamRadio/files/13196265/ic705zno.ini.txt)

armel, 2023-10-28:
I tested with the ic705zno.ini file.
Good news. It's perfectly valid.

So please, try again :

1 - restart your M5Stack
2 - don't care about binLoader step, you can bypass it by pushing left or right button
3 - about iniLoader step, push middle button, choose your ini file in the list with left and right button and select it by pushing middle button.

It must work. If not, https://github.com/armel/M5HamRadio/#help-me-to-debug-your-problem

Armel.

closed 2023-10-29 by TommyMartin