armel/M5HamRadio #62: Will not read my .ini file

Renegade62, 2023-11-10:
Just tried deploying this..
On startup reports Ini file does not exist. Failed to load config file.
It can clearly see it, as the trace log from the burner app displays :

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
----------
Test 40 Mhz
SD/IC705BT.ini
----------
File index 1
/.backup.ini

and thats the lot.

So it can see the IC705BT.ini file.
The file was created with Notepad (also tried Notepad++ with view all symbols to ensure nothing other than text).
The file content is just the example bluetooth for 705 with the mac address changed to suit my 705.
SD Card is 16GB SanDisk Ultra formatted (full not quick) with Fat32.  

; Icom Config
[icom]
icom_model = 705
icom_address = 0xA4
icom_connect = BT
icom_device = 30:31:7D:34:0A:EA

; Wifi Config
[wifi]
wifi_ssid = YOUR_SSID
wifi_password = YOUR_PASSWORD

; Proxy Config
[proxy]
baud_rate = 115200
proxy_url = http://192.168.1.32
proxy_port = 1234

; Transverter Config
[transverter]
transverter_lo_1 = 116000000
transverter_lo_2 = 118000000
transverter_lo_3 = 404000000
transverter_lo_4 = 406000000
transverter_lo_5 = 9968000000

; Geolocation Config
[geolocation]
latitude = 48.848285
longitude = 2.2708201

Any suggestions pls?

Thanks Tim. G6GEI

armel, 2023-11-10:
Hi Tim, 

Thank you for your message.

You need to select the ini file you want to load...

According to the documentation (https://github.com/armel/M5HamRadio/blob/main/README.md#ini-loader) when you see the Ini Loader with a blue gauge, you have 3 seconds to use the bottom middle button to select an .ini file that you put on your SD. So select the .ini file you would like to load and... enjoy.

Armel F4HWN

Renegade62, 2023-11-10:
That works thank you. I had incorrectly assumed the ini file selection was only required if there was more than one.

Tim

armel, 2023-11-10:
Ok Tim, I understand, no problem.

I'm happy that it works now.

Note that, now you have select a valid .ini file with the Ini Loader, a copy was made on the Flash memory of the M5Stack. So if, at the next reboot, you do not make any action / selection of .ini file at the Ini Loader step, it is this last valid .ini file, in Flash memory, that will be read automatically. This way, you can remove the SD card from your M5Stack, if you wish.

Have fun with my small M5HamRadio project ;)

73'

Armel F4HWN.

closed 2023-11-16 by armel