armel/M5HamRadio #65: Which emulator is used for Safari screenshots in README?

zipeijun, 2023-11-18:
This project is awesome!!! but what interests me more is the simulator in the README, which is very helpful for development, which simulator is used?

armel, 2023-11-18:
Hello,

Thank you for your message. In fact, it's not a simulator, but an embedded feature I've developed for my applications. This feature activates a small http server that I query from a simple browser. A simple http request to this http server returns a screenshot of the current M5Stack screen.

It's a bit slow, but it works and I can put nice screenshots in my documentations or social network. 

I seem to be the only one doing this. But it's useful and much nicer than a screenshot taken with a smartphone. 

Regards,

Armel F4HWN.

zipeijun, 2023-11-20:
Thank you so much for your reply, This idea has inspired me as well, it worked so perfectly!

Have a good day, 73

closed 2023-11-20 by zipeijun