armel/M5HamRadio #72: Not working  Core2

lforsolini, 2024-01-27:
I ve tried several times to install in a new Core2 but its not working. Not loading and 
 freezes. Any guesses ?

armel, 2024-01-27:
There have been hundreds of Core2 installations since last year without any problems. The M5HamRadio project is the most downloaded "unofficial" firmware on M5Burner. So I'm very surprised and mostly sorry that you're having problems :(

Does the screen light up? Does it display anything? Not working is a bit short... What about console log ?

Which version of Core2 are you using ? You know if you've got a Core2 v1.0 or the very last new one v1.1 ?

lforsolini, 2024-01-28:
Thank you Armel for your fast response. I love this project. I have a M5 grey and its working well. But I ve bought a Core 2 v1.1 and its not working. I' ve tried  [M5HamRadio.bin]  https://github.com/armel/M5HamRadio/blob/main/firmwares/ESP32/M5HamRadio.bin).
I upload to M5 burner and when finish burning, the Core2 freeze. I can't turning off or reset. I need to remove the baterry and then it doesnt turn ON.
Best regards
Fernando

armel, 2024-01-29:
Hello Fernando,

Ok. Well, the only solution I can see would be to order the latest version of Core2 v1.1. Then I could better understand where the problem comes from. I'm running out of time at the moment and it's going to cost me a bit of money, but I'm adding it to my todo list.

73'

Armel F4HWN.

armel, 2024-01-29:
Humm, I have an idea Fernando, stay tuned...

armel, 2024-01-29:
Ok, Fernando. Can you go back to M5Burner and download version v0.1.3 of the M5HamRadio project in the Core2 section and tell me if it works?

73'

Armel F4HWN.

lforsolini, 2024-01-29:
Hi, Armel
Thank you so much. Now its working. It's solved.
I'd like to thank you for your comitment. You are a very nice guy.
If you need something from Brazil, be my guest to ask.

73 my friend.

armel, 2024-01-29:
Oh, thank you Fernando. If you only knew how happy I am to have managed to understand the bug and fix it without having Core2 V1.1. I'm even happier than you are :) So thank you too for your help and test !

w0rld, 2024-03-30:
I have a Core 2.
I have install the USB driver.
I have attached a screensnap to show what I am trying to burn. 
I have run the serial monitor. The error and serial output follow.
******************************
error when trying to upload

--chip auto --port /dev/tty.wchusbserial57130456621 --baud 1500000 --before default_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x000 /Applications/M5Burner.app/Contents/Resources/packages/firmware/9600bd14b647da11b258d9a7cc173f64.bin
xcrun: error: invalid active developer path (/Library/Developer/CommandLineTools), missing xcrun at: /Library/Developer/CommandLineTools/usr/bin/xcrun

Serial monitor

[FROM M5Burner] /dev/tty.wchusbserial57130456621 opened.
SDCard Type = 3 
SDCard Size = 59344 
SDCard Type = 3 
SDCard Size = 59344 
SDCard Type = 3 
SDCard Size = 59344 
SDCard Type = 3 
SDCard Size = 59344 
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[     9][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   457][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
M5Core2 initializing...E (819) ledc: ledc_channel_config(607): gpio_num argument is invalid
[   826][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
OK
[   838][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
[   839][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
biton read 0x5, data = 0x1, write back 0x5
biton read 0x5, data = 0x4, write back 0x5
[   855][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
biton read 0x3F, data = 0x8, write back 0x3F
biton read 0x3F, data = 0x10, write back 0x3F
biton read 0x3F, data = 0x8, write back 0x3F
biton read 0x3F, data = 0x20, write back 0x3F
biton read 0x3F, data = 0x4, write back 0x3F
E (1033) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
biton read 0x3F, data = 0x80, write back 0xBF
biton read 0x35, data = 0x30, write back 0x35
Addr:0x34 - Name:Axp2101
I2C Axp2101 Found
Addr:0x38 - Name:CST Touch
I2C CST Touch Found
Addr:0x68 - Name:IMU6886
I2C IMU6886 Found
Addr:0x51 - Name:BM8563
I2C BM8563 Found
[  4586][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
SDCard Type = 3 
SDCard Size = 59344 
biton read 0x5, data = 0x30, write back 0x35
SDCard Type = 3 
SDCard Size = 59344 
SDCard Type = 3 
SDCard Size = 59344 
SDCard Type = 3 
SDCard Size = 59344 
[FROM M5Burner] Serial Port Disconnected.

w0rld, 2024-03-30:
*****************************************************
Never mind. I reloaded xcode and it burned.
******************************************************

> I have a Core 2. I have install the USB driver. I have attached a screensnap to show what I am trying to burn. I have run the serial monitor. The error and serial output follow.
>
> error when trying [...]
*****************************************************
Never mind. I reloaded xcode and it burned.
******************************************************

armel, 2024-03-30:
Ok, no problem.

closed 2024-03-30 by armel