armel/M5HamRadio #74: question about working

lu4ba, 2024-02-28:
Hi! first of all thanks for all your code and effort. Sorry for my rustic english.

Im trying to configure the M5Stack board, but actually i dont have access to the equipment (Icom IC705). Its mandatory to see any image in the M5 the conectivity to the IC705, or i can made some test before a friend lend me the HF equipment?

Thanks
Matias LU4BA

lu4ba, 2024-02-28:
[FROM M5Burner] COM6 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
E (335) psram: PSRAM ID read error: 0xffffffff
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
----------
Test 40 Mhz
SD/IC705BT.ini
SD/DXTracker.ini
----------
File index 2
/.backup.ini

armel, 2024-02-28:
Hi,

Without IC-705 (or IC-7300 or 9700), you will only see an ICSMeter screen with the message "Need Pairing". So it's totally useless. 

Armel F4HWN.

lu4ba, 2024-02-28:
I saw the DX Cluster map, thanks for the help!!!!