armel/M5HamRadio #75: core 2 rebooting on connect ic705

aurochsen, 2024-06-22:
Hi,
Core 2 reboots on connection to an ic705. Here is the error log - 
`ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
----------
Test 40 Mhz
SD/IC705BT.ini
----------
File index 1
/.backup.ini
wifi-ap
wifi-pass
705
A4
1
30 31 7D 34 D1 40 
115200
http://192.168.1.32
1234
116000000.00
118000000.00
404000000.00
406000000.00
ovf
Wifi Client Connected
Wifi Client Disconnected
Attempt 1 - Make sure IC transceiver is available and in range.Attempt 2 - Make sure IC transceiver is available and in range.BT Client Disconnected
Attempt 3 - Make sure IC transceiver is available and in range.Bluetooth initialized
NEED PAIRING
BT Client Disconnected
BT Client Disconnected
BT Client Disconnected
NEED PAIRING
BT Client DisconnectedNEED PAIRING

BT Client Disconnected
NEED PAIRING
BT Client Disconnected
NEED PAIRING
NEED PAIRING
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4015e496  PS      : 0x00060030  A0      : 0x8015a976  A1      : 0x3ffe6810  
A2      : 0x3ffdad50  A3      : 0x00000000  A4      : 0x3ffda63c  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x80000001  A8      : 0x00000004  A9      : 0x00000000  
A10     : 0x3ffc8bd8  A11     : 0x3ffe6838  A12     : 0x3ffc8bd8  A13     : 0x00060023  
A14     : 0x00060020  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400935a8  LEND    : 0x400935be  LCOUNT  : 0xffffffff  

Backtrace: 0x4015e493:0x3ffe6810 0x4015a973:0x3ffe6830

ELF file SHA256: 39475bd873b4339c

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
----------
Test 40 Mhz
SD/IC705BT.ini
----------
File index 1
/.backup.ini
wifi-ap
wifi-pass
705
A4
1
30 31 7D 34 D1 40 
115200
http://192.168.1.32
1234
116000000.00
118000000.00
404000000.00
406000000.00
ovf
Wifi Client Connected
Wifi Client Disconnected`

armel, 2024-06-22:
Hi,

Thank you for your message.
Your M5Stack is correctly flashed and your .ini file seems correct. That sounds good.
 
Now, about pairing, if your BD Address is the good one (30 31 7D 34 D1 40), proceed like this.

1 - start normally your M5Stack on ICSMeter or ICMultiMeter
2 - on your IC705, goto Menu > Set  > Bluetooth Set > <<Pairing Reception >>

You will see a Connection request on the screen of your IC 705 (sometimes, after many seconds). Just click on Confirm on the IC705 screen. And here we go. If not, there is a problem somewhere.

Armel.

aurochsen, 2024-06-22:
Ah that was it, I was trying to connect instead of waiting for the request. Thanks!

closed 2024-06-22 by aurochsen