armel/M5HamRadio #82: DXTracker Failed to obtain time

HZ1SF, 2024-07-19:
Hello Armel,

I've been having this issue with DXTracker on two Core2's: 
After the ini is loaded I get to barcode screen and I get an IP address then it gets stuck on "Loading datas"
I tried accessing the IP address but it keeps timing out.
Both Core2's are freshly flashed.

Here is the serial monitor output:

`[FROM M5Burner] COM27 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
----------
Test 40 Mhz
SD/DXTracker.ini
----------
File index 1
/.backup.ini
****************
****************
24.67
46.66
Wifi Client Connected
192.168.1.199
Failed to obtain time
`

Thank you in advance!

73 DE HZ1SF

armel, 2024-07-19:
Hello,

Thank you for your message.
Probably a network problem? Have you tried several times?
I've just tested and everything works.

73 de F4HWN

HZ1SF, 2024-07-19:
Hello Armel,

You are right, it is a network issue. I just tried connecting the Core2 to my phone's hotspot and it worked fine, I will see if I can fix my router.

Thanks again and 73
HZ1SF

armel, 2024-07-19:
You're welcome 😌

closed 2024-07-19 by armel