armel/M5HamRadio #86: WiFi instead of BT for IC705?

DcnPat, 2024-08-30:
Thanks for this really great software for the M5!

I was wondering if it would be possible to have the ICSMeter communicate with the IC-705 via WiFi instead of BT. My amplifier connects via BT and I would like to use both the ICSMeter and amp at the same time.

DcnPat, 2024-08-31:
> Hello,
>
> I haven't found any way to send or receive CI-V commands over Wifi. So,
> for the moment, no.
> Finally, thank you for your donation. It's very kind of you.
>
> 73'
>
> Armel F4HWN

Hello,

You are quite welcome for the donation. I wanted to show my appreciation
for your work.

I was hoping since wfview works and has code for sending/receiving ci-v
over tcp/ip it might be possible.

73,

Pat NE4PO
Deacon Patrick Ouellette
Good Shepherd Catholic Church
8710 Mount Vernon Highway
Alexandria, Virginia 22309

armel, 2025-03-06:
I'm closing this issue. I still haven't found a way to send or receive CI-V commands over Wifi. Sorry.

closed 2025-03-06 by armel