armel/M5HamRadio #90: Did not work on M5Stack Tough

Andi3D-tech, 2024-10-02:
Hello and thank you for this great project!!!

I have the image on one M5Stack (tho one with physicaly buttons) and it works great!

Now i want to do the same on the TOUGH version for outdoor activitys.

M5Burner -> CORE2 & TOUGH -> M5HamRadio Core2  v0.1.5 -> Burn sucessfully

After reboot -> Ini file does not exist -> Failed to load config file

I tried different sd cards, but no change.

Also the touch screen did not work.

Put the test image on the M5 Touch and touch woks perfect.

This is the output of the serial monitor from M5Burner:

`[FROM M5Burner] COM5 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13924
ho 0 tail 12 room 4
load:0x40080400,len:3600
entry 0x400805f0
----------
LittleFS/DXTracker.gz
LittleFS/ICKeyer.gz
LittleFS/ICMultiMeter.gz
LittleFS/ICSMeter.gz
----------
Test 40 Mhz
----------
----------
File index 4
Test 10 Mhz
----------
----------
File index 4
----------
Test 40 Mhz
SD/IC-705-BT.ini
SD/DXTracker.ini
----------
File index 2
/.backup.ini`

armel, 2024-10-02:
Hello,

Thank you.

I'm sorry, but I don't own this model of M5Stack and I've never checked whether M5Hamradio could run on it. Besides, the documentation doesn't mention it as supported (see the very top of the README). 

[image]

Armel F4HWN.

Andi3D-tech, 2024-10-03:
Thank you for your answer.
I try to refund it.

closed 2024-10-03 by armel