armel/M5HamRadio #91: CI-V output?

DL2CC, 2024-10-17:
Hi :-)
Could this faboulous project be used to output CI-V to steer my new JUMA amplifier from the IC-705?
It has a standard 3.5mm jack used for CI-V.

Thanks
73
Frank

armel, 2024-10-17:
Hi Frank,

I don't know. 
By the way, it would be hard for me to work on this subject without having the material. 

73'

Armel.

DL2CC, 2024-10-17:
> Von "Armel FAUVEAU" ***@***.******@***.***>>
> An "armel/M5HamRadio" ***@***.******@***.***>>
> Cc "Frank Grossmann DL2CC" ***@***.******@***.***>>; "Author" ***@***.******@***.***>>
> Datum 17.10.2024 18:09:06
> Betreff Re: [armel/M5HamRadio] CI-V output? (Issue #91)

Thanks Armel,

I am not sure what you mean by material, a device that is steered with CI-V?

I am not sure which radios you have, I guess you have an IC-705.
If you do have a second radio, e.g. an IC7300, you could as well use that as a target for CAT.
Basically all I need is the CAT information coming from bluetooth which is maybe already CI-V formatted to be output to a custom CI-V address via a serial output.

HW?
73
Frank

armel, 2024-10-17:
What I'm doing in this project, using an M5Stack, is sending CI-V commands to an Icom station (705, 7300 or 9700) and processing the returns for display on the M5Stack screen. The CI-V commands format (and return) I use are fully documented by Icom. I have no experience beyond that and am not familiar with the JUMA amplifier.

DL2CC, 2024-10-17:
Yes, I fully understand.

All I need is basically a copy of the CI-V datastream coming from the IC-705 and send it to another device.
It doesn't matter whether it is a JUMA amplifier or something else, it just needs to get the working QRG from the radio.
VE2DX is selling his bluetooth hubs for a lot of money which are doing that.

for example CT17B-6BT V2 https://ve2dx.com/ct17b#376d5018-a825-451f-98fa-6a7177a962d9

73
Frank

Datum 17.10.2024 18:36:52

armel, 2024-10-17:
This VE2 made a lot of money reselling my project when it was open source... The resellers here are also complicit (Ham Radio Outlet, DX Engineering, ML&S). It's a disgrace. I'm not interested in this poor guy, his dubious practices and his dubious products. I just feel very sorry that OMs are losing money by trusting him.

DL2CC, 2024-10-17:
Yes I knew you exactly meant him when you wrote about it :-)

What I am saying is that it is probably quite easy to output the received CI-V data.
Basically copy the stream and set a new target CI-V address and output the frequency data.

But unfortunately I wouldn't know how to do it.

73!
Frank

Datum 17.10.2024 19:16:43

closed 2025-03-06 by armel