armel/M5HamRadio #96: Display 13.2 Module

n0hot, 2025-01-25:
Thanks for this great software, Armel. I have 2 M5Stacks -  one always has DXTracker running and the other ICKeyer.

I would like to put DXTracker on my 17" display so I added the Display 13.2 module. It works but I wanted to check with you to see if there are any ini file statements I can add for modifying the display appearance.  The map appears a bit stretched and the text stair-steps. The monitor info shows that it's displaying @1920x1080, 60Hz. 

Thank you... Scott, NØHOT

[image]

armel, 2025-01-25:
Hi Scott,

Thank you for your message.

Unfortunately, no, there's no way to modify this. You have to understand that the M5Stack’s IPS LCD screen uses a 4:3 format (320 x 240 pixels). So, when outputting to an external 16:9 screen (1920 x 1080), it inevitably affects the display.

However, what you can probably do is check the settings on your external screen. There’s maybe an option to switch to a 4:3 format. You’ll probably see black bars on the sides, but the aspect ratio will be better.

73'

Armel, F4HWN

n0hot, 2025-01-25:
Hi Armel,

Excellent idea! I still do have a 4:3 monitor in storage so the aspect ratio will match perfectly. It's also a bit smaller so everything should look a bit crisper. Thanks!  73,   Scott

closed 2025-03-06 by armel