armel/M5HamRadio #97: failed to load ini file

RW-7, 2025-03-05:
Hello, I've been trying for a while in every possible way to get the m5 to find the ini file.

I always get the message that it can't find it.

Not even when I use the pattern from here and write to the SD card without any adjustments.
Tried several cards
format fat and fat32 no success version 0.1.5

armel, 2025-03-06:
Hello,

Please, read again this [section](https://github.com/armel/M5HamRadio?tab=readme-ov-file#usage).

Armel, F4HWN.

RW-7, 2025-03-07:
thank you. I had overlooked that part.

Another question, there is the transverter mode, but this is not sufficient for the qo100, as vfo a and vfo b are different, but even in split mode the 2nd vfo is still missing. Will there be something like that in the multimeter or is this not planned?

There are a few examples, but unfortunately not for the multimeter (looks great). You can also provide me with the source code so that I can perhaps add a point for the qo100

armel, 2025-03-07:
I lack experience in using Oscar 100.

About source code, sorry, but no way. 

VE2DX made a lot of money with my project when it was open source... The resellers here are also complicit (Ham Radio Outlet, DX Engineering, ML&S). As I explained in the README, because of the systematic plundering of my open source projects by a minority of unscrupulous people, taking advantage of my work to make money, I decided **not to open the source code** of these projects, for the moment.

I was insulted, threatened and blacklisted by these same people (resellers, youtubers, etc.). And even if I know that the majority is behind me, that's enough 😔 Sorry. 

Armel, F4HWN

RW-7, 2025-03-07:
I'm sorry to hear that, I don't like this kind of thing.

The QO100 is actually quite simple
With the IC-705 that I have here, you need both VFOs in split mode.
E.g. VFO A on 2m RX with an offset in the display of xxx MHz, which suits the QO100 at e.g. 144.56230MHz + 10344977700Hz
The VFO B is responsible for sending, i.e. TX in the 70cm band, in my case 432.04380 MHz + 1967998900 Hz
The software should therefore not only change the RX frequency in the diplay in QO100 operation, but also set VFO B to the same mode as VFO A via CI-V and adjust the frequency accordingly. That's basically all.
An example project is e.g. https://github.com/pa3ang/MSSTACK-CORE2-QO-100-for-IC-705/tree/main
a part of it also seems to be yours.

it would be great if you could incorporate something like that, because your ICMuliMeter is an outstanding piece of work and looks very nice...
greetings Ralph